bigjosh/Firmware #1: There is a short audio "pop" on powerup.

bigjosh, 2017-06-24:
Seems like we are playing the right station before the pop, and after. 

Need to track down where this comes from - audio pipeline or FM chip?

ZachDunham, 2017-07-07:
According to the amp datasheet (pg 10) the amp passes audio 4uS after it's taken out of shutdown.  Then the output rises based on the value of the bypass cap.  We could try increasing that value to smooth things out. http://www.ti.com/lit/ds/symlink/tpa6203a1.pdf

bigjosh, 2017-09-01:
Confirmed that the pop happens right after we enable this FM chip here...

` set_shadow_reg(REGISTER_02, _BV( REG_02_ENABLE_BIT ) );    // Enable chip `

MUTE already on so that can't help. Hmmm....

pencerw, 2017-09-01:
@bigjosh you mentioned reaching out to Si in your email - did you mean them, or TI (the amp manufacturer)?

bigjosh, 2017-09-01:
Si. This is an issue with the FM_IC chip. You can prove this by adding a delay between when the crystal oscillator stabilizes and when you enable the FM_IC. When you do, you can hear the pop happens when the FM_IC is enabled - even if all other components have been running for several seconds.  This happens even though the chip is not tuned and the audio is muted. 

This branch adds a 2 second delay before and after the ENABLE...

https://github.com/bigjosh/Firmware/commit/62732242b8a1cb96bd9494b63eb8a3ebe52b2fb9

...with this result...

https://www.youtube.com/watch?v=H6s696xIf_o

There might be bits in other registers you could play with to see if anything could reduce or eliminate that click on enable, but I think all hunting the dark since the undocumented. 

Another mitigation strategy might be to try to hide the click at the very beginning of the real audio by getting the chip tuned up *immediately* after the enable takes effect. Since we already know that this time is widely variable among parts, this would probably require actively hammering the chip to see when the enable finished. This will take empirical experiments and testing.

bigjosh, 2017-09-01:
Tested and setting `AHIZEN` before enable has no effect on pop.

pencerw, 2017-09-01:
Roger!

Just talked to the (now no longer at SiLabs) FAE we had dealt with in the past who said it sounded like a current spike on the audio out line, and that limiting current there would help. I put a call into the new FAE to get more info.

Does that diagnosis sound reasonable to you?

bigjosh, 2017-09-01:
It is definately current out on the audio line... Problem is wrong current
at wrong time.

Only current out on the audio line should be Desposito after time lock. :)

Want me to write up a proper problem description with evedence and
 reproduce instructions?

pencerw, 2017-09-01:
👍🏻👍🏻👍🏻 would you want to talk to this FAE or rather I do it?

bigjosh, 2017-09-02:
Happy either way!

pencerw, 2017-09-02:
Will loop you into the email chain momentarily!