bigjosh/Firmware #3: Sleep between diagnostic blinks?

bigjosh, 2017-06-24:
We currently pull about 0.5mA while blinking a diagnostic code (currently only bad EEPROM CRC).

This currently happens for 120 secs before we go to sleep.

We could drop this to a few uA by sleeping between blinks.

Worth the extra code and complexity?

ZachDunham, 2017-07-07:
@bigjosh IMO this seems like a small thing to prioritize and probably not worth the extra code and complexity.

bigjosh, 2017-08-08:
This ended up being easy once the `sleepFor()` function was done. Besides the small power savings, this ended up saving code size since the `sleepFor()` function is reused and replaces several calls to ~delay_ms()`, which expand to relatively large loops.

closed 2017-08-08 by bigjosh