bigjosh/Firmware #5: Normalize LED brightness based on battery voltage

bigjosh, 2017-06-24:
LED should be uniform brightness regardless of battery voltage level. We can do this by detecting the voltage and applying a normalization gamma correction. 

This is especially important when we eliminate the LED current limit resistor since otherwise 100% brightness will be too bright and use too much power.

bigjosh, 2017-08-08:
This is feeling not so important after playing with actual units. The LEDs does get dim as the battery goes dead, but that could be a feature since it pulls less current then - exactly when you want to save power.

Judgement call: is the LED bright enough now when the battery is dead (it is pretty dim), or should we boost it higher in this case?

pencerw, 2017-09-02:
agree, this is pretty low priority. copying it over anyway.

bigjosh, 2017-09-11:
Deemed not worth it.

closed 2017-09-11 by bigjosh